dbuchwald/6502 #5: Add a "how to use this with Ben's stuff" document

danecreekphotography, 2020-03-20:
I suspect several people will come to this board/project while watching Ben Eater's tutorials on YouTube. The board design has variations from what Ben's doing though so nothing from Ben's tutorials will work directly.

Add documentation that explains the differences between Ben's design and this design, and provide specific steps on how to adapt Ben's code to work on this board:

1. Document the difference in address ranges, perhaps with a helpful graphic to compare the two side-by-side
2. Explain briefly why these changes were made
3. Explain how to update the locations in Ben's sample apps to ones that will work with this board

This documentation is clear enough when I can take Ben's code with updated address locations, assemble it using the assembler he uses, and have it work on this board.

dbuchwald, 2020-03-26:
This should now be covered, let me know if further changes are required

danecreekphotography, 2020-03-26:
Yes, this is fixed!

closed 2020-03-26 by dbuchwald