dbuchwald/6502 #15: Redesign PCB to use smaller footprint SMD components

dbuchwald, 2020-03-26:
The whole circuitry could be largely simplified and the board size reduced if certain components are replaced by SMD counterparts

danecreekphotography, 2020-03-27:
Honestly one of the joys of using your PCB was that it *didn't* use SMD components. For noobs like me who don't have a ton of soldering experience it's way less intimidating to have a board that's all through-hole!

BestDelphini, 2020-07-20:
I learned to solder in high school and then in Navy I had to re-learn as the requirements are much different.
I have a lot solder burns from soldering in very small areas and even upside down. I have not used SMD components and like Dan I am glad there is projects that don't need it. Yes, it is smaller and even less problems with speed. I will give it a chance but don't want to. Doing things that you still understand and can have fun putting it together is what I want.

dbuchwald, 2020-07-20:
While I do understand your point about SMD components (and the fact that they are, essentially, more difficult to work with), there are certain aspects you have to take into account:

Availability of THT/PDIP components - unfortunately, when trying to use more modern chips, your options are very limited. Sometimes you can go with PLCC and THT socket for those, but more frequently than not it's not even an option,
Price factor - PCB can be made significantly smaller, when using SMD components, reducing manufacturing price. The components themselves are cheaper sometimes (not always),
SMT assembly is also significantly easier, cheaper and accessible to home user nowadays. Hot air stations of decent quality are comparable to similar soldering irons, there is plenty of material online on how to use them properly,
Last but not least - there is also my own personal "challenge" factor. I don't like doing the same thing twice the same way, I need to challenge myself, learn and develop new skills. Yes, I was intimidated at first when I saw the size of FT230XS, but I did enjoy the achievement when I managed to solder it on, and I now I have access to variety of chips not available in THT option.

The best, to be honest, would be to create two versions of the board: one that uses SMD components and one that doesn't. After schematic for version 2 is done, the project can be forked or enriched with PR. Let's see where it goes :)

BestDelphini, 2020-07-20:
I will give it a chance and I am sure I will come to the same conclusion. I have been playing with WD's W65C02SXB but was very disappointed. I was looking elsewhere when I found your project and I am willing to give it a go.