dbuchwald/6502 #18: Document Windows environment toolchain

dbuchwald, 2020-03-26:


danecreekphotography, 2020-03-27:
By far the easiest way to run this on Windows is to... not run it on Windows. Instead use WSL2 + Ubuntu from the Windows Store. Everything will build out of the box once you clone the repo with no messing around trying to get make, assemblers, etc. running on Windows.

For code editing Visual Studio Code's remote tools make it a dream. From within the Ubuntu instance you can just run VS Code and edit, save, whatever.

Flashing the ROM is done using xgpro on the Windows side. Arduino serial monitoring is also done with Arduino IDE on the Windows side.

This is how I've done every bit of building/hacking/editing with this project and it's perfect.

The only piece missing is command-line ability to transfer via xmodem. It's a minor pain, but still better than flashing the ROM every time.

dbuchwald, 2020-03-28:
For me the winning combination is Cygwin + ExtraPuTTy, but I need to document it.

closed 2020-03-31 by dbuchwald