dbuchwald/6502 #34: Incorrect value for  RV1 in BOM for clock module

Dave-TR3A, 2020-04-07:
The BOM for the clock module lists a value of 10K for RV1 (a Piher PT10-LV10-105 potentiometer).
 
[image]

That must be a typo: the Piher part number is actually for a 1M potentiometer.

With a 10K pot, the astable circuit generates a square wave with frequency between 62 and 481 Hz, which may be too fast for troubleshooting. Changing to a 1M pot yields a frequency range of 0.72 to 481 Hz.

dbuchwald, 2020-04-08:
Thanks for spotting it! I have just corrected it, can you check and let me know if it can be closed?

Dave-TR3A, 2020-04-08:
Hi Dawid:

I just looked on GitHub. Unfortunately, I’m not seeing the change you made.

Here is a screen shot from moments ago. It shows that RV1 on the BOM is
unchanged, and also shows the URL, date and time.

By the way, I changed the 10K potentiometer on the clock for a 1 M
potentiometer. Now I’m able to generate square waves with frequencies
ranging from .67Hz to 472 Hz (as measured on my oscilloscope). That’s just
what I was hoping for.

Nice job on the clock PCB!

Stay healthy.

Best wishes,

Dave MacKay

*From:* Dawid Buchwald <notifications@github.com>
*Sent:* April 8, 2020 3:25 AM
*To:* dbuchwald/6502 <6502@noreply.github.com>
*Cc:* Dave MacKay <dave@mdmackay.ca>; Author <author@noreply.github.com>
*Subject:* Re: [dbuchwald/6502] Incorrect value for RV1 in BOM for clock
module (#34)

Thanks for spotting it! I have just corrected it, can you check and let me
know if it can be closed?

—
<https://github.com/dbuchwald/6502/issues/34#issuecomment-610796642>, or
unsubscribe

Dave-TR3A, 2020-04-08:
And now the change is showing up. Here’s a screenshot showing it:

Well done! Issue closed.

*From:* Dawid Buchwald <notifications@github.com>
*Sent:* April 8, 2020 3:25 AM
*To:* dbuchwald/6502 <6502@noreply.github.com>
*Cc:* Dave MacKay <dave@mdmackay.ca>; Author <author@noreply.github.com>
*Subject:* Re: [dbuchwald/6502] Incorrect value for RV1 in BOM for clock
module (#34)

Thanks for spotting it! I have just corrected it, can you check and let me
know if it can be closed?

—
<https://github.com/dbuchwald/6502/issues/34#issuecomment-610796642>, or
unsubscribe

Dave-TR3A, 2020-04-08:
Clock module BOM updated to show value of RV1 as 1M ohm.

closed 2020-04-08 by Dave-TR3A

dbuchwald, 2020-04-08:
Thanks a lot, stay safe!