dbuchwald/6502 #49: VGA video output

marcpiulachs, 2020-05-12:
Do you plan to add any kind of video output to the computer? You currently have a ATtiny4313 that is only used for the ps/2 keyboard interface that could probably also be used to generate some rudimentary VGA signal https://hackaday.com/2015/12/17/attiny-does-170x240-vga-with-8-colors/

dbuchwald, 2020-05-13:
Hey, thanks for sharing, and yes, I'm aware of the possibility to use ATtiny as VGA signal generator, but there are two issues with it at the moment:

1. Current solution doesn't provide enough video RAM for any reasonable output (keyboard buffer takes up a lot of memory on ATTiny), so I would need to share system RAM between ATTiny and 6502, and while this is perfectly doable, it requires access to certain 6502 pins that are currently "blocked" (as in: pulled up or down on PCB). This basically requires another revision of PCB and this one might actually be used for it:
https://github.com/tonylweil/6502
2. Even if I ever do any kind of VGA interface, I would rather implement that with much simpler and easier to control set of TTL chips, like the Ben's video card. That being said, I don't have that much interest in VGA signal generation, and I definitely don't want to build any kind of primitive game console. What I wanted this thing to be was educational device that demonstrates basics of computer build and OS development. Serial connection is perfectly sufficient for these purposes...

Again - at some point I might actually consider building discrete, TTL based VGA output, but mostly aimed at text mode. Graphics will be always secondary.

Thanks for showing interest in my project!

marcpiulachs, 2020-05-13:
Hi, it makes perfectly sense. Thanks for your great work with this project, it's a great tool for people like me who came from a software background with little experience on hardware that want to learn the basics of a computer and how it works but does not feel confortable with wires and breadboards, also the software is a great complement to Ben's excelent series. Please keep the great work!

dbuchwald, 2020-05-13:
Thanks for your comment; it seems that quite a lot of us share the same background in software development and this might be the first serious EE project. It is definitely the first for me :) 

And this is exactly where I envisioned this project - to be soft of "no frills" intro to hardware and low level software design, building on top of Ben's introduction to 6502 and computer prototyping.

kyleeddy, 2021-08-12:
Not VGA, but definitely authentic - I've had some initial luck using the [TMS9918a board](https://github.com/jblang/TMS9918A) designed by [@jblang](https://www.github.com/jblang). Connection uses pins available on the J7 connect of your v001 board. The board is designed to work with the rc2014-style systems, but works well with some creative use of the jumpers and an added 74HCT04 for timing delays.

Alex-Kw, 2023-01-17:
> Not VGA, but definitely authentic - I've had some initial luck using the [TMS9918a board](https://github.com/jblang/TMS9918A) designed by [@jblang](https://www.github.com/jblang). Connection uses [...]

If we are mentioning the RC2014, there is an HDMI and Composite Pi terminal, and a VGA terminal. The latter is more relevant here and more available anyway. I have the HDMI one, and it connects to the serial bus; I have TX and RX split so that input comes from a serial device, but output goes to the video terminal. It's likely this could be adapted for many microcomputers with serial text output. https://rc2014.co.uk/modules/pi-pico-vga-terminal/

If I get a chance to try it i will report back, as I intend to obtain one for my RC2014 anyway.