dbuchwald/6502 #50: Unable to get ps/2 keyboard input working

jonlorusso, 2020-08-03:
I've flashed my ATTiny4313 with the keyboard_4313 sketch, using an Arduino as ISP:

[image]

I also flashed the rom with`Software/rom/19_keyboard_test`.

When I boot the computer (with a ps2 keyboard plugged in) the lights briefly flash on the keyboard.

It then shows key `4` as being pressed intermittently, but I am unable to type anything.

I checked my solder joints on the ps/2 connector, and they look ok.

dbuchwald, 2020-08-03:
Hey, thanks for reporting this, but I would need more information about the problem:

1. What kind of keyboard are you using (native PS/2 or via connector),
2. How did you flash the ATTiny4313 (I need details on which pin was connected to which pin on the AVR ISP interface),
3. Are you absolutely sure that the keyboard is working fine?

When I get these details I will repeat the same process (including the ATTiny upload) exactly as you did, but with logic analyser connected to the ATTiny pins to see what happens there. Basically this should work "just like that", I have never seen any issues on my build and I know of other builds where it also worked fine. This might be problem with solder joints on the ATTiny or VIA socket. You can do simple continuity tests on ATTiny pins and respective VIA connectors, but let's start with me repeating your procedure.

jonlorusso, 2020-08-03:
1. What kind of keyboard are you using (native PS/2 or via connector),

PS/2 keyboard

2. How did you flash the ATTiny4313 (I need details on which pin was connected to which pin on the AVR ISP interface),

I used an Arduino UNO as ISP (ArduinoISP sketch).  I then connected the Arduino UNO to the 6502 ICSP interface:

5V -> 5V
GND -> GND
Pin 10 -> RESET
Pin 11 -> MOSI
Pin 12 -> MISO
Pin 13 -> SCK

In the Arduino interface I selected "Arduino as ISP", ATTiny4313 as board/chip, and 8 MHz for speed.

3. Are you absolutely sure that the keyboard is working fine?

To be completely honest, I don't know.  I purchased the keyboard specifically for this purpose, so it's brand new, but obviously could be defective.

Thanks!

jonlorusso, 2020-08-03:
> You can do simple continuity tests on ATTiny pins and respective VIA connectors, but let's start with me repeating your procedure.

For what it's worth, everything appears to be wired up correctly (after continuity tests on all the ATTiny pins).

jonlorusso, 2020-08-03:
One additional piece of information (perhaps useless):

The keyboard lights flash again once (briefly) while the text "Keyboard testing program. is displayed on the LCD, but prior to the "Key pressed: 4" and "Key pressed [ENTER]"

dbuchwald, 2020-08-04:
Hey, sorry, I have tried to replicate the issue you are observing, but it still works on my build. These are the steps I took:

1. Checked out master branch
2. Uploaded rom/19_keyboard_test to EEPROM
3. Uploaded keyboard_4313 using Arduino as ISP programmer (the only difference - used Arduino Mega instead of Uno, as I don't own one)
4. Inserted 1MHz oscillator for system clock
5. Connected PS/2 keyboard
6. Run the computer

It detected keyboard correctly and worked fine displaying keys that I pressed.

So, current version of the code should work correctly with the PCB and working keyboard. These are the possible issues that could cause it:

- Incorrect seating of ATTiny4313 or VIA chip
- Failed upload (I actually had to struggle for a bit to upload the sketch, it didn't work immediately, USBasp is much more stable)
- Soldering issue
- Broken keyboard
- Last but not least - I don't even know if this is possible, but maybe the PS/2 connector you use doesn't match the pinout assumed in PCB design?

I would advise using one of these cheap logic analysers (13$ or so) to confirm that the signal coming to ATTiny is correct. You need to connect two lines to pins 2 and 6 - these are DAT/CLK of PS/2 respectively. If this is correct (represents keys pressed on keyboard), then check ATTiny output on pins 12-19 to confirm that correct data is transmitted to VIA.

Sorry I can't help you any further, but it "just works" on my build...

dbuchwald, 2020-08-05:
One last thing - it's actually possible that the fuses on your ATTiny are incorrectly set. Arduino IDE doesn't burn them with the sketch, you need to "Burn Bootloader" first, even if you don't intend to use the bootloader (which, by the way, is not even possible for ATTiny4313). Try this and let me know if it helps, please.

jonlorusso, 2020-08-05:
> you need to "Burn Bootloader" first, even if you don't intend to use the bootloader

🤦‍♂️🤦‍♂️🤦‍♂️

That was it!  Keyboard is now working! Thanks!

dbuchwald, 2020-08-05:
Awesome, I'm glad we managed to find the reason! I will update the documentation to reflect this step.

dbuchwald, 2020-08-05:
Documentation updated

closed 2020-08-05 by dbuchwald