dbuchwald/6502 #53: help compiling .c files

mattmi91, 2021-08-23:
I am unsure how to compile the .c files in the 6502 projects ROM folder im not sure what to use crygwin? cc65? and what commands do I type into these to compile the .C file into an .out file ive been looking on the internet for days and cannot find anything Im looking for so some pointers on how to do this would be nice :)

rprouse, 2021-08-23:
Every project has a `Makefile`, so running `make clean all` should do it for you. It does use cc65, so that needs to be in your path as does GNU Make. I found that Dawid's makefiles don't work well under Windows, so he uses Cygwin. 

Instructions to get setup are here, https://github.com/dbuchwald/6502/blob/master/Software/README.md#setting-up-development-environment-under-windows

mattmi91, 2021-08-23:
I have tried using cygwin and I did install all the packages he said too but not sure what to type to make the .out file? he doesnt how other then just basics on what to do when you first install it which I already did but what do I type in the terminal something like g++ lcdtest.c? tried this and got like 50 errors

rprouse, 2021-08-23:
Did you try typing `make clean all`?

mattmi91, 2021-08-23:
I get this when I type that make: *** No rule to make target 'clean'.  Stop.

mattmi91, 2021-08-23:
ok so I fixed the make file issue but when I do g++ 4bitlcd.s 
$ g++ 4bitlcd.s
4bitlcd.s: Assembler messages:
4bitlcd.s: Warning: end of file not at end of a line; newline inserted
4bitlcd.s:1: Error: unknown pseudo-op: `.setcpu'
4bitlcd.s:2: Error: can't open lcd.inc for reading: No such file or directory
4bitlcd.s:3: Error: unknown pseudo-op: `.segment'
4bitlcd.s:9: Error: unknown pseudo-op: `.code'
4bitlcd.s:12: Error: no such instruction: `jsr _lcd_init'
4bitlcd.s:14: Error: no such instruction: `write_lcd '
4bitlcd.s:16: Error: no such instruction: `ldx '
4bitlcd.s:18: Error: no such instruction: `lda chars_msg,x'
4bitlcd.s:19: Error: no such instruction: `beq end_loop'
4bitlcd.s:20: Error: no such instruction: `jsr _lcd_print_char'
4bitlcd.s:21: Error: no such instruction: `inx '
4bitlcd.s:22: Error: no such instruction: `bra char_loop'
4bitlcd.s:24: Error: no such instruction: `bra end_loop'
4bitlcd.s:26: Error: unknown pseudo-op: `.segment'

rprouse, 2021-08-24:
`make` builds all the files, you don't call `g++`. The file that you are trying to compile is assembly language, not C++, so it is assembled by `cc65`, but you don't have to call that either, `make all` does that for you and puts the output in a build or bin folder.

Just switch to the `Software` directory and type `make clean all`. That's it, nothing more, it will build everything.

rprouse, 2021-08-24:
I should add that you can't build most of the individual files in this repo and get a runnable program. Almost all of them include files in the `common` directory, that is why you need the `Makefile`, to coordinate the building and linking of all the files.

mattmi91, 2021-08-24:
so im in the directory of where that c file is and when I type make all I get already a directory message

mattmi91, 2021-08-24:
im just trying to get the a.out file from these so I can program it into my eeprom

rprouse, 2021-08-24:
As I said,

> Just switch to the Software directory and type make clean all. That's it, nothing more, it will build everything.

Then look in the `build` folder that is created.

mattmi91, 2021-08-24:
did what you said and here is the error that I got 
$ make clean all
rm -rf build
make -C rom/01_nop_fill ADDRESS_MODE=ext all
make[1]: Entering directory '/home/matthew/6502-master/software/rom/01_nop_fill'
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/zeropage.o -l ../../build/common/zeropage.lst ../../common/source/zeropage.s
make[1]: ca65: No such file or directory
make[1]: *** [../../common/makefile:184: ../../build/common/zeropage.o] Error 127
make[1]: Leaving directory '/home/matthew/6502-master/software/rom/01_nop_fill'
make: *** [makefile:76: build/rom/01_nop_fill.ext.bin] Error 2

mattmi91, 2021-08-24:
tried under the first clean all test he mentions in the rom folder
$ make clean all
rm -f ../../build/rom/*.bin \
rm -f ../../build/rom/*.raw \
../../build/rom/01_nop_fill/*.o \
../../build/rom/01_nop_fill/*.lst \
../../build/rom/01_nop_fill/*.s \
../../build/rom/01_nop_fill/*.map \
../../build/common/*.o \
../../build/common/*.lst \
../../build/common/*.s \
../../build/lib/*.lib
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/zeropage.o -l ../../build/common/zeropage.lst ../../common/source/zeropage.s
make: ca65: No such file or directory
make: *** [../../common/makefile:184: ../../build/common/zeropage.o] Error 127

rprouse, 2021-08-24:
1. `make clean` fails if you have not built before and the build directory does not exist. In that case `make all` will do.
2. To solve your error `ca65` needs to be added to your path. Google how to do that in Cygwin, I think you add it to your `.bashrc` file but I don't use Cygwin. If you can type `ca65` at the command line without errors, then `make` will work.

mattmi91, 2021-08-24:
tried adding the path in my system variables
C:\cygwin64\home\matthew\6502-master\Software\ca65
of course did not work -_- this is why I hate dealing with software so irritating lol

rprouse, 2021-08-24:
If you are using cygwin, that isn't a correct Linux path. Try `~/home/matthew/6502-master/Software/ca65`

mattmi91, 2021-08-24:
im using cygwin on windows 10

mattmi91, 2021-08-24:
ok I solved the problem but im not getting a bin file that I need like he describes 
$ make clean all
rm -f ../../build/rom/*.bin \
rm -f ../../build/rom/*.raw \
../../build/rom/01_nop_fill/*.o \
../../build/rom/01_nop_fill/*.lst \
../../build/rom/01_nop_fill/*.s \
../../build/rom/01_nop_fill/*.map \
../../build/common/*.o \
../../build/common/*.lst \
../../build/common/*.s \
../../build/lib/*.lib
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/zeropage.o -l ../../build/common/zeropage.lst ../../common/source/zeropage.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/sysram_map.o -l ../../build/common/sysram_map.lst ../../common/source/sysram_map.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/rom/01_nop_fill/nop_fill.o -l ../../build/rom/01_nop_fill/nop_fill.lst nop_fill.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/syscalls.o -l ../../build/common/syscalls.lst ../../common/source/syscalls.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/acia.o -l ../../build/common/acia.lst ../../common/source/acia.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/blink.o -l ../../build/common/blink.lst ../../common/source/blink.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/core.o -l ../../build/common/core.lst ../../common/source/core.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/keyboard.o -l ../../build/common/keyboard.lst ../../common/source/keyboard.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/lcd.o -l ../../build/common/lcd.lst ../../common/source/lcd.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/lcd4bit.o -l ../../build/common/lcd4bit.lst ../../common/source/lcd4bit.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/modem.o -l ../../build/common/modem.lst ../../common/source/modem.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/string.o -l ../../build/common/string.lst ../../common/source/string.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/utils.o -l ../../build/common/utils.lst ../../common/source/utils.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/via.o -l ../../build/common/via.lst ../../common/source/via.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/via_utils.o -l ../../build/common/via_utils.lst ../../common/source/via_utils.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/tty.o -l ../../build/common/tty.lst ../../common/source/tty.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/parse.o -l ../../build/common/parse.lst ../../common/source/parse.s
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/menu.o -l ../../build/common/menu.lst ../../common/source/menu.s
ar65 r ../../build/lib/common.lib ../../build/common/syscalls.o ../../build/common/acia.o ../../build/common/blink.o ../../build/common/core.o ../../build/common/keyboard.o ../../build/common/lcd.o ../../build/common/lcd4bit.o ../../build/common/modem.o ../../build/common/string.o ../../build/common/utils.o ../../build/common/via.o ../../build/common/via_utils.o ../../build/common/tty.o ../../build/common/parse.o ../../build/common/menu.o
ca65 --cpu 65C02 -Dclock_mode_flag=2  -Dacia_tx_irq=0  -Ddb6502 -I ../../common/include -o ../../build/common/crt0.o -l ../../build/common/crt0.lst ../../common/source/crt0.s
cp -f ../../common/none.lib ../../build/lib/common.c.lib
ar65 r ../../build/lib/common.c.lib ../../build/common/crt0.o
ld65  -C ../../common/firmware.ext.cfg -m ../../build/rom/01_nop_fill/01_nop_fill.ext.map -o ../../build/rom/01_nop_fill.ext.bin ../../build/common/zeropage.o ../../build/common/sysram_map.o ../../build/rom/01_nop_fill/nop_fill.o ../../build/lib/common.lib ../../build/lib/common.c.lib

rprouse, 2021-08-24:
did you look in the folders under build/roms? I saw them being created in the output you sent.

mattmi91, 2021-08-24:
thanks for the help with the advice of getting that make clean all command, the fix was I had to put cc65 exe files inside the cygwin64 folder then it worked it was a lucky guess but I got it :)

rprouse, 2021-08-24:
Excellent, happy 8-bit computing. Can you close this issue? Only you or @dbuchwald can.

[image]

mattmi91, 2021-08-24:
maybe it would help someone else that has similar problem?

rprouse, 2021-08-24:
It may help someone else, but they will still be able to find it in search. It is normal in GitHub to close issues that are resolved or the questions answered.