dbuchwald/6502 #56: Dead?

Phunter1, 2022-07-15:
Seems everything dropped off 2 years ago. Is this project dead? I would rather not put time into something that is dead.

Kevin-smurf, 2022-07-15:
Hi Phunter1,  I built dbuchwald's project a couple of years ago, after watching Ben Eaters excellent tutorials on YouTube. 
I firstly built Bens 6502 breadboard project after sourcing all the components myself.
Then I found Mr Buchwald's PCB version, and built that too.  I'm not a programming buff, so got a bit lost, but finally got it working.  I learnt heaps from doing this. 
It's been on the "back burner" since then !!!

dbuchwald, 2022-07-15:
Hey, the project is not dead, but indeed, it has been hibernating for last two years. I started redesign of the board, but scope creep and too ambitious goals prevented me from reaching next milestone. Seeing how it blocked my progress I went back to drawing board, decided to split the project in smaller chunks and one of the prerequisites was using latest line of AVR microprocessors with different programming interface.
Long story short, I went to implement the new interface in avrdude project and spent there next six months. Changed jobs in the meantime and, well, life simply happened. The fact that there are other better designs out there doesn't help either, but the main reason for putting that aside was simply lack of interest from the community. Most of these projects end up like that, because we learn from each other and the most persistent prevail. People would rather design their own build, and use others as an inspiration/guideline.

As for putting time into something that is dead, I'm not entirely sure you mean. The PCB has been built by many people, and as far as I know, each of these builds is running fine. Software is usable, and there is enough of it to use it to start building your own applications. So yeah, it's not like you will spend weeks building it and figuring out compilation options - it will simply build, load and run, and if you use PCB you are almost guaranteed to have the thing running in a week.
If you have any issues, you can always contact me here or drop me an e-mail.

Phunter1, 2022-07-15:
> Hey, the project is not dead, but indeed, it has been hibernating for last two years. I started redesign of the board, but scope creep and too ambitious goals prevented me from reaching next [...]

This is the first time I've gotten into projects like this so I may be looking at it in the wrong light.

Nice to know it's not forgotten.

I'll be building this project shortly.

closed 2022-07-15 by Phunter1